positron96/cnc3018-offline-controller #1: Original firmware backup

Bonnee, 2021-10-02:
I want to try this firmware on my offline controller that came with an STM32 clone, but st-info reports 0kb of flash size. I'm guessing the flash is locked and I would need to unlock it, but I want to make a backup of the original firmware before flashing yours.
Did your controller come with locked flash?

positron96, 2021-10-04:
Hi. Nope, I had no problem reading the flash, though I used a J-link programmer for that.

In case you need it, I've saved my firmware, it's in the `docs` folder.
https://github.com/positron96/cnc3018-offline-controller/blob/main/docs/original.hex

Bonnee, 2021-10-04:
I tried with an st-link clone. Maybe the STM32 clone is not compatible with it, but I doubt it. I'll try with a CMSIS-DAP when I get access to one and report back.

positron96, 2021-10-05:
I'm using an st-link clone + openocd for most of the development, used J-link only for the first several days, when I dumped the firmware.

closed 2021-12-13 by positron96