positron96/cnc3018-offline-controller #2: [Question] Another offline-contorller from AliExpress

maxl95, 2021-10-20:
Hey guys,

I also recently bought a 3018 CNC. An offline controller was also included. But this one has an STM32F103RBT6. That means I shouldn't try this firmware, right?

regards maxl95

maxl95, 2021-10-20:
Thank you for your quick response. Yes, you're right, it has a color display. The firmware on the Controller is very bad. I thought I could improve it with your firmware. Unfortunately the read-out protection is activated. So I can't make a backup and make firmware myself.

closed 2021-11-07 by positron96

Jempiken, 2021-11-13:
Hello Max195.. I have the same as yours...  I bricked mine, trying to upload the firmware... from positron
Any chance that I could have the firmware for it.. ??
I first read it though, but When i read it with ST-linkv2  the memory starts at 0x08000000 for 0x3F54 Bytes
It is a lot less bytes.. than the original.hex from positron.
Can you try to read it..
The SW settings is slow speed at 480kHz and the it connects OK ! but only reading from that adress
Tnx 4 help
JP

Jempiken, 2021-11-13:
I t can be read, probably I read too little...  i had to read the full data range, but did not..
I used a white ST-LINK V2
JP

zolo2hin, 2022-09-01:
@Jempiken. @maxl95 guys have you any updates with that? May be we should try to make some branch and start to do from scratch?