positron96/cnc3018-offline-controller #5: No communication after baudrate is set

vinpinto, 2022-01-09:
Hardware:
-Offline Controller based on SMT32F (same as shown in the developer pictures)
-CNC Controller is woodpecker v3.3 with grbl firmware v1.1

What Works:
-Serial probing cycle for baudrate seems to work
-Controller shows interface,.
-All menu entries are shown and can be selected.

What doesn't work:
-The only serial output from the offline controller is the $I sent in different baud rates to probe for the correct baud rate.

Test made:
-With a serial probe in tx rx terminals to check for comms, the behavior is as follows:
-A few $I are sent by the offline controller in different baud rates.
-The machine responses with [VER:1.1f.20170801:] [OPT:V,15,128] ok
-The offline controller understands the response and stops sending any further comunication.

The offline controller only shows the machine as X idle or sometimes Lockpad Idle, but no further comand comes out of the offline controller, not with a SD card file, not with the DRO, or any other entry on the menu.

positron96, 2022-01-10:
Hi, thanks for the report. 

"X" as a status is shown then the firmware thinks it's not connected to grbl. The firmware does not send any commands in this case. So we need to understand why it thinks so. 
The USB port on the offline controller is used as a debug serial port. Please connect a PC to it, and open serial monitor to that port, you will get more info about why it thinks it's disconnected.

Also, Lockpad is shown when the grbl board reports that it itself is connected to PC via USB. It does via the 8th pin on 8-pin ribbon cable. In this case UART cannot be used to supply commands to grbl. At least, my board does this; if the lock shows randomly, then your board is different and you'll need to remove this functionality.

vinpinto, 2022-01-10:
Thanks man, that got everything solved, I'm still not sure why both states X and padlock are dependent on this, but removing this feature made everything work

closed 2022-01-10 by vinpinto

positron96, 2022-01-11:
Ok, I also don't quite understand what happened. Reopen the issue if the problem manifests again, and we'll dig into it.

mppkll, 2022-08-09:
I am having the same issue with the same offline controller (ST32 based, same as you project photos) and the same CNC board (woodpecker cnc v3.3).

Looking at the schematics for Woodpecker CNC v3.3 it does not look like there is any connection to pin 8 of the offline controller jack

1 - 5v
2 - Gnd
3 - 5v
4 - Gnd
5 - TX
6 - Rst
7 - RX
8 - NC

When I connect the USB to the computer and use Putty to watch the out put I get
```
 > (f 15,128) ''(len 0) 
 > (f 15,128) 'Grbl 1.1h ['$' for help]'(len 24) 
Msg '[MSG:'$H'|'$X' to unlock]'
 > (f 15,128) '[MSG:'$H'|'$X' to unlock'(len 25) 
GrblDRO::onButton(1,1)
GrblDRO::onButton(1,0)
GrblDRO::onButton(0,1)
GrblDRO::onButton(0,0)
GrblDRO::onButton(5,1)
GrblDRO::onButton(5,0)
GrblDRO::onButton(6,1)
GrblDRO::onButton(6,0)
GrblDRO::onButton(3,1)
GrblDRO::onButton(3,0)
GrblDRO::onButton(2,1)
GrblDRO::onButton(2,0)
```
Pressing the buttons will show as above but pressing [Exit/Step], selecting Reset (Ctrl-X), and [OK/Spindle] does not show any output   on the Putty screen.

I tried connecting a UART directly to the offline controller and when I sent "[VER:1.1f.20170801:]" after I saw "$I" in the terminal window the display will change to the "X IDLE" screen but pressing any buttons does not get any output.  I tried sending the "[OPT:V,15,128]" and "ok" as well but it did not change the behavior.

Above vinpinto "Thanks man, that got everything solved, I'm still not sure why both states X and padlock are dependent on this, but removing this feature made everything work"  Do you know what he removed to make it work?  Any other thing I might look at?

vinpinto, 2022-08-09:
Hello @mppkll , I had to force the USB to PC connection to false and got everything working, tomorrow I can have a look and send you the exact code modification

positron96, 2022-08-10:
Yes guys, if your board has an unconnected 8th pin, then PC connection detection needs to be removed from code as it will detect PC as connected and prevent the controller to send anything. 
However, if you see "X" and not lockpad, then there might be something else going on which I not totally understand, probably baudrate autodetection works incorrectly.

positron96, 2022-08-10:
Actually, can anyone test this fix: add a "connected = true;" to GCodeDevice.h GCodeDevice::begin (around line 44) method so that it looks like this:
```
    virtual void begin() { 
        while(printerSerial->available()>0) printerSerial->read(); 
        connected = true;
    };
```

I don't remember why I've removed it, but at the moment I can't see that flag being set to true anywhere.

vinpinto, 2022-08-10:
I had it looked and this is what a I changed:
From file Watchedserial.h

    bool isLocked(bool forceRead=false) {
        /*if(forceRead || int32_t(millis()-lastUpdate) > (int32_t)CACHE_DURATION) {
            cachedLocked = digitalRead(pin)==HIGH;
            lastUpdate = millis();
        }*/
        return false;
    }

mppkll, 2022-08-10:
@positron96 Thanks for the info that makes it clear what was going on hardware wise.  I tried the code change you suggested and that gave me a lock icon rather than an X but still did not function.  Then I added the code change that @vinpinto and that worked.

Thank you both very much and additional thanks @positron96  for making this in the first place!