positron96/cnc3018-offline-controller #6: smaller controller

wolfgangbc, 2022-03-07:
Hello,

great job with this firmware. Thank you.

Just an Information:
My most recent GRBL Offline Controller has no real STM Chip on it, but an GD32F103C8T6.
This chip has only 64k, so the verify step of the programmer fails because the code is too large to fit.

I plan to replace the chip by a STM32F103 with 128k, as soon as they are available again.

wolfgangbc, 2022-03-07:
> Thanks for the info.
>
> It might still be possible to fit the firmware into 64k, it's not much larger than that. Some revisions ago it still fitted in 64k, but I had to use custom float printf and scanf

I am sure, you tried „optimize for size“ compiler option.

stonemull, 2022-03-08:
Just confirming mine has the same controller as ops, GD32F103C8T6.

stonemull, 2022-03-31:
No I unplugged the controller entirely and fitted an esp8266 in its place,  installed a virtual serial port driver on the computer and run the CNC over wifi instead currently.

TWill974, 2022-03-31:
Ok ! Unplug the chip is not easy.... For me, the easy way, will be to fit the firmware.... Maybe, wolfgangbc or positron96 have do it....????

Srhitchcock, 2022-07-09:
Why will it not work with the gd32 chips?

positron96, 2024-04-12:
For anyone still interested, GD32 clones of STM32F103 seem to really have 64k instead of unofficial 128k of STM32F103 (e.g. look [here](https://www.eevblog.com/forum/projects/do-you-have-an-stm32f103c*t6-looking-for-your-help!/msg1672862/#msg1672862)). This firmware is larger than 64k, I could not make it smaller. So to support GD32, it should probably be rewritten without Arduino core (which uses a lot of flash). Hope this project will help anyone who decides to do this.

closed 2024-04-12 by positron96

andrzejolszak, 2024-08-25:
For anybody getting the "Programming error @ 0x08010000!" and considering what to do: instead of installing an old version consider instead stripping out the USB support from the latest master. It's simple to do, you just need to remove any code mentioning USB, remove USB-related build parameters, and revert two commits that changed 64k->128k - this gets the size well below the 64k. The serial ribbon cable connection will still work normally, so if you were not planning to use the USB anyway, then no harm done. The alternative is to go back deep into version history for a <64k version, but that one is missing a lot of important features of this firmware.

positron96, 2024-08-29:
@andrzejolszak nice catch! Indeed USB stack takes a lot of space, so if you don't need debug logging, it can be safely removed.