positron96/cnc3018-offline-controller #7: Impossible to flash GD32F103C8T6 chip

TWill974, 2022-03-30:
Hi,

When i flash the .hex firmware with  a st link usb key v2, all seems to be OK but at the verification step , i have this error :  Programming error @ 0x08010000!

Is it due to the problem of short memory of the GD32F103C8T6 chip ?

Do you have the same firmware but lighter ?

positron96, 2022-04-04:
Hi. I believe this is a duplicate of #6 and yes, AFAIK GD chip has really 64k, and the firmware is larger that 64k. Actually I tried to keep it under 64k before giving up, so if I roll back some months of commits, I might get a sub-64k firmware. It will lack features though.

closed 2022-04-04 by positron96

TWill974, 2022-04-09:
Hi,
OK i understand...If you can try to delete, the less interesting features ...But the best will be to have a light firmware with all features.... :-)  Please, let me know, when you will generate the light firmware < 64k

Thanks a lot.