positron96/cnc3018-offline-controller #9: Love the firmware but

Almazick, 2022-05-18:
It doesn't work on bigger files for some reason, but it works with smaller files.  What could be a problem?  When it tries to change z-axis I have to press unlock to continue.

Almazick, 2022-05-24:
I tested with Fusion 360 and had the same problem.

positron96, 2022-05-24:
Is it possible to share the gcode file?

Almazick, 2022-05-24:
[program.zip](https://github.com/positron96/cnc3018-offline-controller/files/8759345/program.zip)

Here you go.  Let me know what seems to be a problem

positron96, 2023-02-03:
Hi. Sorry for (very) long response. I've looked at the gcode and have found anything suspicious. I haven't had access to my machine yet, but hopefully will soon. Meanwhile, do you remember whether any gcode worked at all for you?

dturner, 2024-02-07:
Unfortunately, I'm having the same problem. The following GCode fails 2/3 times, seemingly at random points in the file. All I see on the controller is "error: 22". Could it be that an interrupt (maybe for screen updates) is stopping GRBL from processing the GCode at a critical time? 

[inlay-problem.ngc.zip](https://github.com/positron96/cnc3018-offline-controller/files/14187348/inlay-problem.ngc.zip)

Happy to help debug this. Would you mind sharing how to set up the `GD_DEBUGF` macro to print debug information?

positron96, 2024-02-07:
Hi! Does the gcode work if you send it to grbl from the PC? I just googled "vcarve grbl error 22", and it seems people have this problem from time to time, which means it could be related to PC software/post-processor, not FW.

Apart from that, I think GD_DEBUGF is already enabled by default, and will print info to SerialUSB. Connect the controller (not grbl board) to PC via mini-USB cable, it should be detected as COM port, so viewable from Arduino IDE, for instance. 

Might be wise to power both the PC and CNC from one socket, as there is no galvanic isolation. Or it could be that USB-powered controller will power grbl board as well, even if you don't supply power to CNC. I don't quite remember.

dturner, 2024-02-07:
Thanks for the quick response. Yes, the GCode works if I send it from my Mac. I'm using Universal GCode Sender (UGS). 

When sending the same GCode from the offline controller it seems to fail at the 50% more often than not (line 55 in the GCode file). Here's the serial output when that happens: 

```
  J queueing line 'G1 X14.973 Y2.437 Z-1.309', len 25
 > (f  9,  4) '<Run|MPos:-275.170,-163.806,-4.600|FS:300,10000>'(len 48) 
 > (f 10, 22) 'ok'(len 2) 
<  (f 10, 22) '?' RT
 > (f 10, 22) '<Run|MPos:-275.610,-166.269,-4.600|FS:300,10000>'(len 48) 
<  (f 10, 22) '?' RT
 > (f 10, 22) '<Run|MPos:-276.568,-168.563,-4.600|FS:300,10000>'(len 48) 
<  (f 10, 22) '?' RT
 > (f 10, 22) '<Run|MPos:-278.371,-170.266,-4.600|FS:300,10000>'(len 48) 
<  (f 10, 22) '?' RT
 > (f 10, 22) '<Run|MPos:-280.750,-170.935,-4.600|FS:300,10000>'(len 48) 
<  (f 10, 22) '?' RT
 > (f 10, 22) '<Idle|MPos:-281.986,-172.489,-4.600|FS:0,10000>'(len 47) 
<  (f 10, 22) '?' RT
 > (f 10, 22) '<Idle|MPos:-281.986,-172.489,-4.600|FS:0,10000>'(len 47) 
<  (f 10, 22) '?' RT
 > (f 10, 22) '<Idle|MPos:-281.986,-172.489,-4.600|FS:0,10000>'(len 47) 
<  (f 10, 22) '?' RT
 > (f 10, 22) '<Idle|MPos:-281.986,-172.489,-4.600|FS:0,10000>'(len 47) 
<  (f 10, 22) '?' RT
 > (f 10, 22) '<Idle|MPos:-281.986,-172.489,-4.600|FS:0,10000>'(len 47) 
```

At this point the servos have stopped and the spindle is spinning at max rpm. The only thing which stops it is sending a reset command via the offline controller menu.  

Any suggestions on how to debug this further?

positron96, 2024-02-07:
Ok, I tried this file on my board, and I notice some problems with communication between grbl and stm32. Going to investigate.

dturner, 2024-02-07:
Just a quick update. I tried stripping whitespace from the GCode and was able to get more successful runs, roughly 50% of the time it would complete successfully. 

The fact that it fails at random points in the file would indicate a race condition or electrical interference.

dturner, 2024-02-08:
@positron96 Thanks for investigating. What problems did you notice? Maybe I can investigate as well.

positron96, 2024-02-08:
Well, if you look in debug stream when the file is being sent, there are less "ok" responses (from grbl) than there are sent g-codes (from stm32). I use [character counting](https://github.com/gnea/grbl/wiki/Grbl-v1.1-Interface#streaming-protocol-character-counting-recommended-with-reservation) to send the data, so the algorithm relies on grbl confirming that all commands have been received. With missed "ok"s at some point the counter cannot send anything anymore unless you reset the boards (from menu), at which point you start getting error:22 or error:36 from grbl

positron96, 2024-02-08:
@dturner could you pleased try the latest version to see it it fixes the problem for you?
@Almazick sorry for forgetting about this problem for so long. If it's still relevant, hopefully this fixes the problem for you as well.

dturner, 2024-02-09:
That fixed it! I ran the same GCode 10 times without a single failure - thanks very much for looking into this and fixing the problem. 

Out of interest what made you think it was the serial receive buffer size that was causing the issue?

positron96, 2024-02-09:
So, I made a UART sniffer out of Arduino Mega that captures data from both RX and TX lines and sends them to PC. That showed that "ok" responses are there in correct quantity, and they are lost in stm32 firmware. Then I logged `Serial.available()` values in the firmware and realized that this value approached the size of the RX buffer, which is 64 bytes by default. So UART data from grbl overflowed the buffer, and extra data was silently ignored.

I am going to close this issue, but feel free to add feedback.

closed 2024-02-09 by positron96

dturner, 2024-02-09:
Thanks for the explanation and the fix 👍 .