positron96/cnc3018-offline-controller #13: Impossible to flash APM32F103RBT6 chip

RepRapThailand, 2023-02-03:
Hi,

 My hardware is APM32F103RBT6.

[image]
[image]
[image]

Thank you.

positron96, 2023-02-03:
Hi. I've never heard of this chip and how to flash it. Also, I am afraid this project does not support your hardware, as it seem to contain a full-color TFT display (and a microSD slot which usually indicates a newer hardware with a TFT display). So you are on your own, sorry.

RepRapThailand, 2023-02-04:
Hi, 
for chip
https://www.sunrom.com/p/apm32f103rbt6-lqfp64-geehy-arm-cortex-m3-replacement-for-stm32f103rbt6
my display is OLED.

positron96, 2023-02-04:
Thank you for the info. However, I have no experience with this chip, so can only give limited advice. Actually your MCU should have 128kb officially, while the original STM32F103 only had 64k officially, that might be the difference in settings. You could probably compile the firmware for STM32F103RBT6 and try to upload that.

Also, you'd better check that schematic of your board matches one used in the firmware. As the SD slot is different on your board, there might be other differences.

closed 2024-04-12 by positron96