positron96/cnc3018-offline-controller #16: Unable to Connect

RDWC-Canada, 2024-01-17:
I'm trying to install this firmware on my STM32F103C8T6 based offline controller. I've installed all drivers for the St-Link V2 (Clone) updated its firmware and verified the pins are correct. I've soldered the SWD connection point wires, and installed Platformio within VS Code. I've cloned the files and compiled them. When I hit upload I get "init mode failed (unable to connect to target)" Do I have to delete the original firmware first? Or am I missing a reset process?

positron96, 2024-01-17:
Hi! Here are some thoughts:
1) Did you power the board? I think some chinese STLinks do provide power to the board, others do not, and by standard they should not. So powering the board by USB cable may be required in your case (preferably from the same PC to have the same GND levels)
2) Do you have other STM32 devices, a blue pill maybe? Then you can try to connect to it first using as much of your cabling as possible to check if cables work.
3) Does openocd say anything useful in addition to "init mode failed (unable to connect to target)"? 

Replacing original firmware and resetting the board programmatically still requires a working SWD connection, so no, you are not missing these steps.

RDWC-Canada, 2024-01-17:
Hi! I appreciate the help! 

1) I've tried supplying power through the SWD pads in both 3.3v and 5v from the ST-Link dongle. Now I've also tried providing power via USB from the PC. (I disconnected GND and VCC from the dongle first) In all instances the original firmware loads properly and I can navigate through the screens like normal. 

2) I only have this one STM32 based device and an ESP32 based project. Unfortunately the ESP32 doesn't support SWD. 

3) Heres what I get at the end of the upload attempt :
___________________________________________________________________________________
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
debug_level: 1

0x20000
hla_swd
Error: init mode failed (unable to connect to the target)
in procedure 'program'
** OpenOCD init failed **
shutdown command invoked
_______________________________________________________________________________________

I have checked my solder connections with a microscope, no cold solder joints, no bridging between pads. I'm not sure what else there is to try.

positron96, 2024-01-17:
Ok, nothing interesting in openocd output.
STLink will not work if its GND is not connected, and it should not work if its VCC is not connected. So even if you power the controller board with USB, all 4 wires need to be connected to STLink for it to work.
When you only connect 4 wires from STLink and nothing else, does the original firmware start up?

positron96, 2024-01-17:
One other thing is to check whether your STLink is properly recognized by openocd. Unfortunately, I don't know how to check this. At least, there should be no unknown usb devices in Windows device manager.

RDWC-Canada, 2024-01-17:
Success! I connected to the alternate 3.3v supply on the STLink dongle (it has two 3.3v pins) and the "New and Improved" firmware has now been properly installed. not sure why that fix was the solution though, since both pins had 3.3v when tested.

Side note: I also managed to cancel my Amazon order in time, so its a win-win.

Thanks for all the help!

closed 2024-01-18 by positron96