positron96/cnc3018-offline-controller #21: can this fit on the c8 version of the chip?

revjmoney, 2024-07-24:
i screwed up and flashed this firmware to my controller without looking at the chip first and mine has the c8 variant of the chip and platformio says it's 12000 bytes too big.  my controller is bricked for now unless i can figure out how to fit this?  maybe take out stuff?  all i need is jog and dro really.  i wish i had enough coding skills to fix this!  thanks for your time anyways!

revjmoney, 2024-07-24:
so i realized i may be missing something but my stlink wont program this and my thingy is bricked.

revjmoney, 2024-07-24:
ok, i have a gd32f103c8t6 on my board.  idk what this means, i just want the thing to turn back on. :)

positron96, 2024-07-25:
Hi. So this is a duplicate of #7 and unfortunately only original STM32 C8T6s have 128k, chinese copies like GD32 or APM32 have 64k. You can reflash the original firmware, the one from my machine is in README

closed 2024-07-25 by positron96

revjmoney, 2024-07-25:
i also could not burn your original firmware file... i found another on here (github) somewhere that worked! ty